Harmonus/HMS_SlicerModule #7: Better communication of accuracy information

nicoleaucoin, 2017-02-06:
Revisit the way that the user is told how accurately the biopsy site was positioned with respect to the target.

jcmagill, 2017-02-06:
Nicole, please take me off of this list.

pmarathakis, 2017-02-06:
Sorry John

> Revisit the way that the user is told how accurately the biopsy site was
> positioned with respect to the target.
>
> <https://github.com/Harmonus/HMS_SlicerModule/issues/7>, or mute the
> thread
auth/AMzOXP2a2QOvzs5xJ-DXDRT_8LmjF51Zks5rZz1dgaJpZM4L4VwL>

jcmagill, 2017-02-06:
No problem. How are things?

nicoleaucoin, 2017-02-06:
Looks like you're off the team, but are you still following the repos?

Nicole

jcmagill, 2017-02-06:
Yes.

pmarathakis, 2017-02-06:
Going pretty good, but crazy busy. We are in amigo now doing not really
official phantom tests. We're getting really good accuracy too, averaging
around 1-1.5mm.
We have a new software engineer, who's worked on Slicer for like 10 years
now and she's making things work how it's meant to.
Are you still getting these notifications? I checked the github and you
aren't on any of the teams or anything.
How's things going with you?

nicoleaucoin, 2017-02-06:
@jcmagill You're listed as watching this repo, that will result in getting the notifications, you'll have to unwatch yourself.

jcmagill, 2017-02-06:
Glad you are getting good results.  Did you go back to FDA yet?

Things here are not great. I came in to Draeger in the middle of a conflict
between my boss and his counterpart about, among other things, what my role
should be and who I should report to.  My boss was suddenly fired two weeks
ago.  I am secure for now but it is not a great environment. I am hoping
things get better now that the tension at higher levels is reduced.

The central feature of every professional endeavor is conflict.

John

> No problem. How are things?
>
>
> > Sorry John
> >
> >
> > Nicole, please take me off of this list.
> >
> >
> >
> > > Revisit the way that the user is told how accurately the biopsy site
> was
> > > positioned with respect to the target.
> > >
> > > <https://github.com/Harmonus/HMS_SlicerModule/issues/7>, or mute the
> > > thread
> > auth/AMzOXP2a2QOvzs5xJ-DXDRT_8LmjF51Zks5rZz1dgaJpZM4L4VwL>
> >
> > >
> >
> > <https://github.com/Harmonus/HMS_SlicerModule/issues/7#
> > issuecomment-277718661>,
> > ceUqSYXQlUVSw8oRks5rZz3LgaJpZM4L4VwL>
> >
> > <https://github.com/Harmonus/HMS_SlicerModule/issues/7#
> issuecomment-277719018>,
> auth/AMzOXEV7omz3X_KSIKlvmg_NTBJIE2C4ks5rZz4bgaJpZM4L4VwL>

> <https://github.com/Harmonus/HMS_SlicerModule/issues/7#
issuecomment-277720044>,
b2jGFWAIFICDpo62KeqCWiVlFyNks5rZz7tgaJpZM4L4VwL>

closed 2017-09-07 by nicoleaucoin