Harmonus/HMS_SlicerModule #18: Password protection of sensitive files

pmarathakis, 2017-02-07:
We will want to save DICOMS and procedure information for resuming.
These files will need to password protected on windows to not allow access.

nicoleaucoin, 2017-04-03:
Even if not implementing resume functionality, password protect anything we save that may have the patient name in it etc

nicoleaucoin, 2017-06-06:
Look at setting a password on the dicom database (one overloaded open call takes a password):
http://doc.qt.io/qt-4.8/qsqldatabase.html

nicoleaucoin, 2017-07-31:
Log files also need passwords.

nicoleaucoin, 2017-08-15:
https://wiki.qt.io/User_Login_Dialog

nicoleaucoin, 2017-09-06:
Hide folders:
https://www.windowscentral.com/how-hide-files-and-folders-windows-10

nicoleaucoin, 2017-09-15:
From another system:
zip the files with a password [at the end of the patient]

nicoleaucoin, 2017-10-03:
zlib from Slicer, no password option?

nicoleaucoin, 2017-10-03:
Adding another third party option:
http://quazip.sourceforge.net/index.html

nicoleaucoin, 2017-10-05:
ctkDICOMDatabase can call database() and get QSqlDatabase
    (that supports setPassword) and open(pword)
    - except can't get at it from python
slicer.dicomDatabase.database()
no attribute

Libs/MRML/Core/vtkMRMLTableSQLiteStorageNode.cxx: has a passwrod for writing a table node to an sqlite db

Can I make my own db and set a password on it? Would have to check for instances of slicer.dicomDatabase or else reset it to mine.