Harmonus/HMS_SlicerModule #75: Self check at start up - risk mitigation

nicoleaucoin, 2017-07-18:
Add Risk analysis checks to start up check.

nicoleaucoin, 2017-08-30:
Disk space check available in Qt5. Look at QSystemStorageInfo  vs QStorageInfo

pmarathakis, 2017-08-30:
To calculate checksum values, download FCIV from https://support.microsoft.com/en-us/help/841290/availability-and-description-of-the-file-checksum-integrity-verifier-u

To run checksum calculator:
pathtoFCIV/FCIV - md5 pathtofile/file.exe
to get checksum for a file
or
pathtoFCIV/FCIV - md5 pathtofolder
to get checksum for all files in a folder

nicoleaucoin, 2017-08-30:
Free space:
if platform.system() == 'Windows':
        free_bytes = ctypes.c_ulonglong(0)
        ctypes.windll.kernel32.GetDiskFreeSpaceExW(ctypes.c_wchar_p(dirname), None, None, ctypes.pointer(free_bytes))
        return free_bytes.value / 1024 / 1024

nicoleaucoin, 2017-08-30:
Do the python compiled files.

nicoleaucoin, 2017-08-31:
Merged.

closed 2017-08-31 by nicoleaucoin