Harmonus/HMS_SlicerModule #124: Allow setting the Windows date/time

nicoleaucoin, 2017-08-17:
Since the machine may not be on a network with time updates, allow the users to reset the date time without requiring the top level admin password.

Add a Settings menu.
Pop up the Windows date/time setting interface.

Related to https://github.com/Harmonus/HMS_SlicerModule/issues/40

pmarathakis, 2017-10-13:
Also show the time in ProBx please

nicoleaucoin, 2017-11-20:
Good for synching with the MR console clock.

nicoleaucoin, 2017-11-21:
Showing clock:
http://doc.qt.io/qt-4.8/qt-widgets-digitalclock-example.html

To set the clock can exit to windows and restart, but can also call
SetSystemTime
https://msdn.microsoft.com/en-gb/library/windows/desktop/ms724942(v=vs.85).aspx

Populate a clock via QTimer, right click to set system time?