Harmonus/HMS_SlicerModule #127: Active slice viewer shows latest loaded scan by default

pmarathakis, 2017-08-17:
When working on a scan (such as in targeting or most commonly confirmation), when a new scan is sent to the software and automatically loaded, the active slice viewer shows the new scan by default. This is not desired as it could disrupt what you are currently working on

nicoleaucoin, 2017-08-18:
Comment from Peter:
Can we preserve zoom and window level settings as new confirmation scans come in?

nicoleaucoin, 2017-09-28:
From the Slicer list, can turn off volume propagation with a flag on the SliceCompositeNode to turn this off on a per-slice view basis. 
https://github.com/Slicer/Slicer/blob/703557e658c976f7d146ff4c8408ef43ec7dea00/Libs/MRML/Logic/vtkMRMLApplicationLogic.cxx#L124

nicoleaucoin, 2017-09-29:
Slicer pull request that gives a show option:
https://github.com/Slicer/Slicer/pull/800

closed 2017-11-13 by nicoleaucoin