Harmonus/HMS_SlicerModule #224: Check DICOM direction requires a better method

pmarathakis, 2017-10-18:
The method for checking DICOM direction is flawed.
The new method should be something like this:

step 1) normalize the direction vectors:

Rvector[3] = first 3 values of DICOM direction
Avector[3] = last (next) 3 values of DICOM direction
Rvector[3] = Rvector/((Rvector[0]^2+Rvector[1]^2+Rvector[2]^2)^0.5)
Avector[3] = Avector/((Avector[0]^2+Avector[1]^2+Avector[2]^2)^0.5)

step 2) work out distance between expected vector and DICOM vector

Rexpected = [1,0,0];
Aexpected = [0,1,0];
Rdist = ((Rvector[0] - Rexpected[0])^2+(Rvector[1] - Rexpected[1])^2+(Rvector[2] - Rexpected[2])^2)^0.5;
Adist = ((Avector[0] - Aexpected[0])^2+(Avector[1] - Aexpected[1])^2+(Avector[2] - Aexpected[2])^2)^0.5;

step 3) check that Rdist and Adist are < epsilon

nicoleaucoin, 2017-11-08:
@pmarathakis first pass done, will extend the testing.

nicoleaucoin, 2017-11-13:
Merged

closed 2017-11-13 by nicoleaucoin